Ma7moud3ly/micro-repl #2: This device is not supported!  Right now we support only Micro...

davefes, 2023-04-14:
I believe I am using an OTG adapter as a mouse works on my Samsung Galaxy A03s.

I am supplying an external 5V to my ESP32 DevkitC.

Do I need another adapter?

Thanks.

Wind-stormger, 2023-04-14:
Same issue.

SONY XQ-AT72, Android 12

ESP32S3 micropython 1.19.1, REPL output from USB , ❌

ESP32S2 micropython 1.19.1, REPL output from USB , ❌

ESP32S2 micropython 1.19.1, REPL output from UART , ❌

Ma7moud3ly, 2023-04-14:
Hello @davefes @Wind-stormger 
Thank you for testing the app.
Actually this issue happens because I made an incorrect filter for devices by vendor Id.. 
```
    private val supportedVendors = listOf(
        11914 //Raspberry Pi (Trading) Limited
    )
```
I thought 11914 related to MicroPython itself not Raspberry Pi Pico 😅

So I will find another filtering method other than the vendor id. 
Thanks

Wind-stormger, 2023-04-17:
@Ma7moud3ly Is it ready for esp32？

Ma7moud3ly, 2023-04-17:
@Wind-stormger It should be ready.
Please test the new release and let me know.

Ma7moud3ly, 2023-04-17:
@FakeChimera01 Have you tried V1.1 ?

Ma7moud3ly, 2023-04-17:
@FakeChimera01 
For filtering MicroPython boards I need some help from those who have different boards.
I asked here for help but didn't get any response.
https://github.com/orgs/micropython/discussions/11272

Wind-stormger, 2023-04-18:
Me too, V1.1 not work on ESP32S2 or ESP32S3.

Ma7moud3ly, 2023-04-19:
Hello guys,
In the new version, I have added an option to approve any port manually and add it to the white list.
All your boards are expected to be working now..
Thanks for your help.
[image]

for update -> https://play.google.com/store/apps/details?id=micro.repl.ma7moud3ly

closed 2023-04-19 by Ma7moud3ly

davefes, 2023-04-19:
Working for me on ESP32.  Very nice job.  The ESP32 dev board is even powered from my phone so can program on the go!

Syntax highlighting is probably a big ask.

Something odd happens when you rotate the phone, you lose the connection ... I am confident that it is not my data cable connections.