Ma7moud3ly/micro-repl #3: Lose connection if rotating the phone.

davefes, 2023-04-19:
Working for me on ESP32. Very nice job. The ESP32 dev board is even powered from my phone so can program on the go!

Syntax highlighting is probably a big ask.

Something odd happens when you rotate the phone, you lose the connection ... I am confident that it is not my data cable connections.

Turning rotate off and it works. For the editor it would be nice to go to landscape :)

Ma7moud3ly, 2023-04-20:
Hello @davefes 
The app's behavior on screen rotation and landscape mode is a good catch and will be handled in the next release/s.
Regarding the ESP32 dev board, I'm not sure you can test it and tell us your experience 🤔
And really the code editor needs a lot of work to be a code editor. It might be replaced with any open-source editor if this was possible.
Thanks.

davefes, 2023-04-20:
I can run a program.  Took awhile to work out {} runs the interpreter, after searching how I could do a CTRL-C, CTRL-D and CTRL-X on my Samsung A03s :)

Ma7moud3ly, 2024-04-29:
- Should be working with several Micropython ports
- A new code editor was added in version 1.3 based on [sora-editor](https://github.com/Rosemoe/sora-editor) 
- Screen orientation will be disabled in the next release

closed 2024-04-29 by Ma7moud3ly