Ma7moud3ly/micro-repl #7: F-Droid inclusion of Micro REPL

emdete, 2023-08-14:
I found your app "Micro REPL" and want to add it to the f-droid repo. The app contains some dependencies that collide with the rules given by f-droid so i removed them but did not change anything else.

Please let me know if  you are fine with the inclusion.

licaon-kter, 2023-08-14:
ref: https://gitlab.com/fdroid/fdroiddata/-/merge_requests/13564

linsui, 2023-10-01:
Could you please tag releases so that we can update the app automatically?

Ma7moud3ly, 2023-11-29:
Tags will be added to releases starting from v1.4

closed 2023-11-29 by Ma7moud3ly

licaon-kter, 2023-11-29:
Builds fine if I downgrade to Java17: https://gitlab.com/fdroid/fdroiddata/-/merge_requests/13564/diffs?commit_id=801ce4632962e80954d965affca30963da9ecd55

...because Debian has Java17 or 19 in the future (Testing now)

@Ma7moud3ly are any 18 features used?

licaon-kter, 2023-11-29:
Regarding reproducibility ( https://f-droid.org/docs/Inclusion_How-To/#reproducible-builds ) would have been nice, but you currently use Firebase/Crashlytics.

If you'd switch to a FOSS one like ACRA or Bugsnag or Sentry, we could have tried repro.

Ma7moud3ly, 2023-11-29:
- It is acceptable to either upgrade or downgrade Java version, and the application will build without encountering any issues.
- Regarding Firebase services, I am uncertain about finding an alternative, but I can add a new build type that excludes Firebase dependencies from the built apk, however they will still in the source code. Users can then build the app using the following commands:
``` ./gradlew assembleRelease``` and ```./gradlew assembleGmsRelease```.

licaon-kter, 2023-12-04:
@Ma7moud3ly gradle flavor without Firebase sounds good, now we're doing the clean up before building https://gitlab.com/fdroid/fdroiddata/-/merge_requests/13564/diffs#9f7512cb773fbef9421c237b1f999d2a50e523d7_0_25

ping us when ready to test

Ma7moud3ly, 2023-12-05:
@licaon-kter Now we have two build flavours ```gms``` with firebase services and ```default``` without.

licaon-kter, 2023-12-05:
@Ma7moud3ly builds fine, any chance to build an APK after merging https://github.com/Ma7moud3ly/micro-repl/pull/11 from `default` flavor and put it in https://github.com/Ma7moud3ly/micro-repl/releases/tag/v1.4 so we can check for reproducibility?

Ma7moud3ly, 2023-12-07:
@licaon-kter done ✅

licaon-kter, 2023-12-07:
Ah forgot about java version :(

But that's the last of our worries, looks like a new case of https://gitlab.com/fdroid/fdroiddata/-/issues/3138

difflog: 
[micro9.log](https://github.com/Ma7moud3ly/micro-repl/files/13603158/micro9.log)

linsui, 2023-12-17:
Please update Compose Compiler to 1.5.6. See https://gitlab.com/fdroid/fdroiddata/-/issues/3138

Ma7moud3ly, 2023-12-18:
@linsui done ✅

licaon-kter, 2023-12-19:
thanks @emdete ands @linsui 

Now, if only Fast lane ( https://gitlab.com/-/snippets/1895688 ) folder&files can be setup with description, icon, summary, pics, we are ready to merge