Ma7moud3ly/micro-repl #19: Circuit python

nikulinleo, 2024-03-06:
With Circuit Python MCU is only visible and only works terminal
If I try opening explorer there is no files, and if I try to create one, app crashes
Found with esp32c3 and esp32s2, both works correctly with PC

Ma7moud3ly, 2024-05-07:
CircuitPython has a different files system than  MicroPython. I will work to support it soon
Thanks

stanelie, 2025-09-26:
I would also like to be able to updates CircuitPython files on my devices from micro-repl on the go.

Thanks!

Novan24, 2026-10-14:
> With Circuit Python MCU is only visible and only works terminal If I try opening explorer there is no files, and if I try to create one, app crashes Found with esp32c3 and esp32s2, both works correctly with PC

Same on Raspberry pi pico too
i don't have PC, need it to use display library