Ma7moud3ly/micro-repl #20: Explorer and save issues.

handsomedude, 2024-04-16:
App works great so far.

Open explorer without Pico connected, get phone storage.
Connect Pico and open explorer and you get Pico.
Fair enough, bit you can't switch storage back to phone to load a script while Pico is connected.

If you select a script from Pico storage, you cannot copy and save it to phone.  

Open script from Pico into editor, disconnect the Pico and you still have no option to copy to pc.

Your save button should do same as thorny app, ask where you want to save, Pico or this computer .

Also, if you select explorer option and Pico is connected, you should have an option to explore Pico or phone

JustaLiriK, 2025-04-21:
I could copy a script from another editor and paste and save it to the esp32 once.
Now i have trouble adding another script.
I do need two scripts as modules to fullfill my project.
Any clue?

Ma7moud3ly, 2025-05-12:
@JustaLiriK What’s the issue with the other script? You can either create one using the code editor or upload it from your phone storage and then import it into your main script.

Ma7moud3ly, 2025-05-13:
@JustaLiriK kindly check this https://github.com/Ma7moud3ly/micro-repl/issues/44#issuecomment-2874670827

closed 2025-05-13 by Ma7moud3ly

JustaLiriK, 2025-05-13:
> @JustaLiriK kindly check this https://github.com/Ma7moud3ly/micro-repl/issues/44#issuecomment-2874670827

Hello, 
Didn't see that issue,i'm gonna try this workaround. I guess i'd rather rename my first main module "main_old" before importing the second one.
Thanks.