Ma7moud3ly/micro-repl #23: F-Droid can't build

licaon-kter, 2024-05-08:
https://gitlab.com/fdroid/fdroiddata/-/jobs/6807565850#L1093

you tagged https://github.com/Ma7moud3ly/micro-repl/releases/tag/v1.5 from commit https://github.com/Ma7moud3ly/micro-repl/commit/2ed4b6e33eb6276addbca855fb3f39e143e512aa

but APK was built from https://github.com/Ma7moud3ly/micro-repl/commit/5fb7a842a3378b84541ff01865d464807c667456 :(

Ma7moud3ly, 2024-05-13:
I hope v1.6 will build without any issues🙌
Thanks

licaon-kter, 2024-05-14:
Please reopen, 1.6 is not repro: [micro11.log](https://github.com/Ma7moud3ly/micro-repl/files/15307020/micro11.log)

linsui, 2024-10-17:
1.7 build failed. https://github.com/Ma7moud3ly/micro-repl/commit/cc1646074db2431b509ba98b291345d4a940097b set the jdk version to 18. Could you please use 17?

Ma7moud3ly, 2024-10-18:
@linsui done ✅

linsui, 2024-10-20:
It's reproducible finally. But the key is changed. What happened?

Ma7moud3ly, 2024-10-20:
@linsui Do you mean the signing key for build ?
Yes it is, I lost the old one and changed it on Google play 🥲

linsui, 2024-10-20:
Oh, no... OK, please send an email from m.aly@gasable.com to team@f-droid.org for verification. :)

Ma7moud3ly, 2024-10-20:
@licaon-kter **9182817**

linsui, 2024-10-21:
Updated.