Ma7moud3ly/micro-repl #25: App crashes when opening file from scripts (when not connected to board)

Afnankabiro, 2024-05-11:
App keeps crashing when trying to view files from scripts while a board is not connected. Haven't tried with a board connected yet

Device: Realme 7 Pro
OS: Android 11

Screen recording below

https://github.com/Ma7moud3ly/micro-repl/assets/102663023/7eb73f9d-ac40-4bc7-8fdd-9a51d493f0c4

Afnankabiro, 2024-05-11:
> App keeps crashing when trying to view files from scripts while a board is not connected. Haven't tried with a board connected yet
> 
> Device: Realme 7 Pro OS: Android 11
> 
> Screen recording below
> 
>  Record_2024-05-11-17-11-58.mp4

Afnankabiro, 2024-05-11:
Still crashes when connected to board (ESP32)

Recording below-

https://github.com/Ma7moud3ly/micro-repl/assets/102663023/2d7e6799-b4f2-4e86-b362-3dd6102228d2

Afnankabiro, 2024-05-11:
Can't get into the editor. App crashes 

Recording below-

https://github.com/Ma7moud3ly/micro-repl/assets/102663023/bbad25d2-ff15-4c28-8487-0d52c19409b3

Ma7moud3ly, 2024-05-12:
Hello @Afnankabiro,
There was a problem in release 1.5 with the code editor.
It has been fixed and 1.6 will be released soon 
Thanks.

closed 2024-05-12 by Ma7moud3ly