Ma7moud3ly/micro-repl #27: App crashes lately when trying to edit code

janihorvath, 2024-05-14:
Hi,

Thanks for the app, very promising!

It used to work fine but lately every time I try to edit any code on the RP2040 Zero, the app crashes...

What is wrong? Could you please fix it?

Thank you!

Ma7moud3ly, 2024-05-14:
Hello @janihorvath
code editor crashing has been fixed in version 1.6.
Please wait for the update on Google play or download it from releases here.
Thanks

closed 2024-05-14 by Ma7moud3ly

janihorvath, 2024-05-14:
Hello Mahmoud,

thanks for the swift reply. Unfortunately I cannot install from github, my
phone is my corporate's, it is restricted...
When can I expect 1.6 on Google Store?

Thank you!

Mahmoud Aly ***@***.***> ezt írta (időpont: 2024. máj. 14.,
Ke 18:38):

Ma7moud3ly, 2024-05-15:
Hello @janihorvath 
Version 1.6 is available now on google play, please test and let me know if there are any issues.
Thanks