Ma7moud3ly/micro-repl #28: Filesystem stays even after resetting??

Afnankabiro, 2024-05-17:
When resting board form the app, it's filesystem seems to stay intact. Am I doing something wrong?

Device : Realme 7 pro
Board: ESP32

Video below-

https://github.com/Ma7moud3ly/micro-repl/assets/102663023/5240f012-8066-44eb-aca6-ba1cd36cd165

Afnankabiro, 2024-05-17:
Pressed the reset button on my ESP32. The file is still there but it seems to be empty tho...

Video below-

https://github.com/Ma7moud3ly/micro-repl/assets/102663023/6ae6ad06-2d14-48bd-a1c4-6fd9a28fd087

Ma7moud3ly, 2024-10-16:
Thank you for reporting this, 
Actually `reset` doesn't mean to wipe the flash memory at all,
- reset calls machine.reset()
```Resets the device in a manner similar to pushing the external RESET button.```
- machine.soft_reset()
```Performs a soft reset of the interpreter, deleting all Python objects and resetting the Python heap.```
- Terminate =  Ctrl-C 
```cancels any input, or interrupts the currently running code or loops.```
Maybe reset or  soft_reset don't exist for ESP32, so they have no a function
https://docs.micropython.org/en/latest/library/machine.html#reset-related-functions

closed 2024-10-16 by Ma7moud3ly