Ma7moud3ly/micro-repl #37: Remove `DependencyInfoBlock`

licaon-kter, 2024-11-08:
ref: https://gitlab.com/fdroid/fdroiddata/-/jobs/8310542226#L553

We find that there is a `DependencyInfoBlock` in your APK.

It's a [Signing block](https://bi-zone.medium.com/easter-egg-in-apk-files-what-is-frosting-f356aa9f4d1) added by AGP and encrypted with the Google public key so it can't be read by anyone else except Google. You can read more about it [here](https://gitlab.com/fdroid/admin/-/issues/367) and [here](https://gitlab.com/fdroid/fdroidserver/-/issues/1056).

While this was added a while ago, we were only enforcing it for new apps, and recently we started scanning updates too.

Could you please disable it with the following code?

```
android {
    dependenciesInfo {
        // Disables dependency metadata when building APKs.
        includeInApk = false
        // Disables dependency metadata when building Android App Bundles.
        includeInBundle = false
    }
}
```

Thanks!

Ma7moud3ly, 2024-11-11:
Hello @licaon-kter 
I have included ```dependenciesInfo```  at  ```default``` flavor but app size still the same.. Is it ok with you ?
```
create("default") {
     dimension = "services"
      dependenciesInfo {
           // Disables dependency metadata when building APKs.
           includeInApk = false
           // Disables dependency metadata when building Android App Bundles.
           includeInBundle = false
      }
 }
```

licaon-kter, 2024-11-11:
The block is small, will check soon :)

licaon-kter, 2024-11-12:
@Ma7moud3ly ok, so when is the next release coming? :)

Ma7moud3ly, 2025-02-07:
It should have been fixed at V1.9
Thanks in advance 😊

closed 2025-02-07 by Ma7moud3ly