Ma7moud3ly/micro-repl #41: Can't open serial port and board is not in File Manager

DJP728, 2025-03-29:
I'm using a pico 1 connected via a USB data cable to a USB C to A adapter, and another A to micro adapter. For some reason I can't open the board in File Explorer and my device sends me a notification to fix the issue by formatting even though it does nothing.  Also get a "Cant open serial port" error in Micro-Repl when I press approve after pressing "connect". I'm using a Galaxy s21 with the newest version of Android. The Pico is definitely flashed with MicroPython.

DJP728, 2025-04-15:
@Ma7moud3ly I've lost my USB C->A adapter, so I can't see if it is still broken. I've not tried with an older device, and all the other ones I could use are relatively modern. I recently (few weeks ago, few days before I opened this issue) flashed the firmware for Micropython, so it's probably a recent version. I'll check later though.

3dgoose, 2025-04-15:
> [@DJP728](https://github.com/DJP728) Do you still face this issue ? Have you tried with another device with older version of android ? Also can you mention the MicroPython version ? Thanks

Sometimes. Now theres a new issue : when you put the pico in a certain position it doesnt detect it.
(are you tunisian ?)

Ma7moud3ly, 2025-04-16:
Hello @3dgoose the issue seems to be with your device or cable, not the app.
(Nope I'm from 🇪🇬)

closed 2025-05-11 by Ma7moud3ly