Ma7moud3ly/micro-repl #47: Uploading script not working

VE2ZAZ, 2025-07-10:
Hi!
I've just installed Micro REPL on both a Motorola G7 phone and a Samsung tablet, and am seeing the same issue. I am trying to upload (click on up-arrow) a micropython script to a Raspberry Pi Pico running Micropython v 1.25.0 (2025-04-15) using a USB_On-The-Go cable. The app says that the file was saved (as a pop-up message), however I do not see the file show up in the explorer. Deleting a file and creating a directory works, but uploading a file does not. I have sourced the file from two locations (Google Drive and phone/tablet flash storage) without success. Using the Pico on a PC with Thonny via USB all works fine. Any idea what I may be doing wrong? 
Thanks.

VE2ZAZ, 2025-07-23:
No reply, so I bounce my request back up. Can you give me some indication of what I may be doing wrong? Thanks. Bert.

Ma7moud3ly, 2026-09-04:
Hi Mr.Bert,

Sorry for the very late response, and thank you for your patience. I’m not actively maintaining the app as often as I’d like.
Could you please try the latest release? This issue may have already been fixed ...
If you’re still experiencing it, could you let me know the size of the script? That would help me to reproduce the issue.

Thanks again for reporting this and for your detailed testing 🙏

VE2ZAZ, 2026-09-10:
Hi. 
Thanks for following up on the issue. Better late than never!

I have tested the upload with the latest micro REPL version (September 4 2026) and the issue persists. I cannot upload a file. The Raspberry Pi Pico is recognized, the app shows me its directory content (currently only one file called boot.py). When I try to add another .py file, it says that the file was "saved to /". However the file never shows up in the "MicroPython~/" directory. I have checked with a PC and the file is simply not there. The script file I am trying to transfer is 15.93 KB in size.

One other gripe I have is that we do not know where the "Local Scripts" directory is (the one shown in the Scripts function). When I search the on tablet for anything related to this, I just cannot find that location. So I cannot upload any locally edited scripts to the Pico, only downloaded scripts from the internet (from Google Drive, for example). From within the Scripts function, there should be a way to upload the listed files to the target device, but there is not. You can run them, but you cannot transfer them. Uploads can only be performed from the Explorer function. And since I do not know where the scripts are saved, I cannot transfer any of the created scripts to the Pico.

There you go. Thanks again for your help. It is a very nice app, it just needs a bit of tweaking.

Bert.

sijan246, 2026-09-11:
What is the size of the file? I think 10 KB or above, right? The same problem happens with mine even though I have tried several times after updating the app. It works fine if the uploaded file is kept under 5 to 9 KB

sijan246, 2026-09-11:
not just me it is the porblem of many people . so please fix this

VE2ZAZ, 2026-09-11:
@sijan246 Yes, as I said, it is 15.93 KB. I will try with a smaller file. Thanks for the insight.

VE2ZAZ, 2026-09-11:
@Ma7moud3ly OK, confirmed that a small file will transfer correctly. I tried a 2 KB file and it worked.
Please fix! 
Thanks.
Bert.

earthmarch23-bit, 2026-09-13:
Hi, adding another data point to this thread.

- Board: ESP32-S3 (ESP32 GENERIC S3)
- MicroPython version: v1.28.0 (build 2026-04-06)
- App version: Micro REPL 3.1
- File I'm trying to upload: ina219.py, a sensor driver library, 
  exactly 15.0 KB (15002 bytes)
- Result: same as reported above — app shows the file as "saved", 
  but it never appears in the device's file explorer afterward

This matches @sijan246's observation that files under ~9 KB work 
fine but anything larger consistently fails. Confirming the issue 
is still present on the latest release (3.1).

Hope this extra data point helps pin down the cause — happy to 
test a build if you have one.

concretedog, 2026-09-15:
I've just flashed a Pico with micropython version 1.28 and I have Micro Repl version 3.1 running on android. I find I have the same issue above but have a workaround. If I open the explorer I am in a "Micropython" directory and if I make a folder in there and then add a python file using the "add new" button it creates the file on the RP2040 device. If I try this via the editor directly, so I open the editor and then try and save the file to the device I cannot, I can write into a file and run the file, but can't save to the device unless I do the workaround I mention.

closed 2026-10-02 by Ma7moud3ly